Esri/cordova-plugin-advanced-geolocation #1: i get error

roysG, 2016-03-23:
Uncaught ReferenceError: AdvancedGeolocation is not defined(…)

it is hapenning when i try to run the code:

 AdvancedGeolocation.start(function(success){

        try{
            var jsonObject = JSON.parse(success);
            console.log("aa:",jsonObject);
        }
        catch(exc){
            console.log("Invalid JSON: " + exc);
        }
    },
    function(error){
        console.log("ERROR! " + JSON.stringify(error));
    },
    ////////////////////////////////////////////
    //
    // REQUIRED:
    // These are required Configuration options!
    // See API Reference for additional details.
    //
    ////////////////////////////////////////////
    {
        "minTime":500,         // Min time interval between updates (ms)
        "minDistance":1,       // Min distance between updates (meters)
        "noWarn":true,         // Native location provider warnings
        "providers":"all",     // Return GPS and NETWORK locations
        "useCache":true,       // Return GPS and NETWORK cached locations
        "satelliteData":false, // Return of GPS satellite info
        "buffer":false,        // Buffer location data
        "bufferSize":0         // Max elements in buffer
    });

(The code taken from your exapmple)

andygup, 2016-03-23:
Hi, that error typically means one of two things. Make sure you are running AdvancedGeolocation after the [`deviceready`](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/sample/map.js#L15) event. Also make sure the plugin is installed using the CLI (Command Line Interface) command `cordova plugin ls`.

roysG, 2016-03-23:
i ran the command and i found the plugin as installed:

"cordova-plugin-advanced-geolocation 0.1.0 "Cordova Advanced Geolocation Plugin - Android"

I also run this command after device is ready.

What can be the problem ?

andygup, 2016-03-23:
The error means the plugin isn't visible within the scope of your Javascript code. It's not a problem with the plugin, itself.

Have you tried running the sample app included with this repo? Instructions are in the README.

roysG, 2016-03-23:
i try to running this code from debugger chrome console

andygup, 2016-03-23:
I also recommend hooking up your phone to the Android Studio native debugger when doing Cordova development. The debug messages in the Android Studio console might provide other important clues as to what's going on.

roysG, 2016-03-23:
i also tried implement the code inside my project and call the function, but same error.

My system info:

Your system information:

Cordova CLI: 5.3.3
Ionic Version: 1.2.4-nightly-1917
Ionic CLI Version: 1.7.8
Ionic App Lib Version: 0.6.4
ios-deploy version: 1.7.0
ios-sim version: 5.0.1
OS: Mac OS X Yosemite
Node Version: v4.2.2
Xcode version: Xcode 6.4 Build version 6E35b

andygup, 2016-03-23:
**Recommendation 1** - confirm the Android version as well as the make/model? This plugin only works with Android.
**Recommendation 2** - Install [Android Studio](http://developer.android.com/tools/studio/index.html) and make sure you can see the phone's debugger output. I recommend this anyway as a best practice for doing Cordova development. This is the "only" way to get native debugger information from the phone and can be very useful in troubleshooting issues.
**Recommendation 3** - follow the Quick Start recommendations in the [README](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/README.md), create a new test application and see if that works. If it doesn't work then please copy and paste the Android Studio debugger output here.

roysG, 2016-03-23:
Look at my image i attach, and notice to plugin "rateUs" 

They added their plugin to window scope, and i identity this plugin in easy.

All the plugins i installed i can find them in window scope, except to your plugin.
Maybe you miss something :( ?

FYI.

andygup, 2016-03-23:
Did you complete all three of my recommendations above and if so what were the results for all three? 

>>All the plugins that i was install can be find in the window scope, except to your plugin.

The most likely reason for this is the plugin isn't starting up correctly. The best way to figure this out is to follow my three recommendations and report back on the answers. 

Make sure you are running the app on an Android phone (v5 or newer) and **not an emulator.** This plugin will not work on iOS.

roysG, 2016-03-24:
i will do it and i let you know.

roysG, 2016-03-25:
I tried to do what you ask by filter all the log cat to  your package name, but i get no log for that.

adb logcat | grep "com.esri.cordova.geolocation.AdvancedGeolocation"

I also test the adb logcat without filter your package and i got messages.

Please let me know what to do.

andygup, 2016-03-25:
1) Please confirm here your model of Android phone and the operating system version?
2) Let me know if you are using the sample app that comes with this repo?
3) Copy and paste all unfiltered logcat messages starting from the geolocation application startup to when it finished loading into a .txt file and attach it here. 

Let's see what info that provides. Thank you!

roysG, 2016-03-25:
The demp app working for me, i do not know why it is not working in my app, very strange.

Second thing, i was thinking i can get the cells id, can you upgrade the plugin to get all cells id around the device?

This will be awesome!

andygup, 2016-03-25:
>>Second thing, i was thinking i can get the cells id, can you upgrade the plugin to get all cells id around the device?

Great suggestion. I opened an enhancement request [here](https://github.com/Esri/cordova-plugin-advanced-geolocation/issues/2) to investigate.

andygup, 2016-07-07:
This issue was closed back in March via https://github.com/Esri/cordova-plugin-advanced-geolocation/commit/057228f4fd1d12fc79d2ab72464d88489e23a8f4

closed 2016-07-07 by andygup

fusionstream, 2017-06-11:
Sorry for opening a closed issue but I have the same issue. I'm on phoneGap and am new to it. There appears to be 2 copies of the same plugin in ../plugins and ../node_modules relative to my www folder. I'm not particularly new to Javascript so it does sound like a scope issue. I'm just not sure why.

Aside from the folders, this was added to my config.xml
<plugin name="cordova-plugin-advanced-geolocation" spec="https://github.com/esri/cordova-plugin-advanced-geolocation.git" />

I can't directly link to the js file because it's outside of my root.

Any ideas on how I can fix it? I'm not that far into building this app. Should I just switch to Cordova?

andygup, 2017-06-13:
@fusionstream yes, there should be two copies as you described. I'm only using Cordova, so I'm not able to answer PhoneGap specific questions. But in general for this library it shouldn't make a difference as long as you are using the latest version.

When the plugin is installed correctly it will be available under the `window` global namespace. See screenshot below. 

This project's [`plugin.xml`](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/plugin.xml) file describes everything that gets copied over to the native android project.

My recommendations are:
1) Make sure you are running the latest Cordova
2) Follow the instructions in the [Quick Start](https://github.com/Esri/cordova-plugin-advanced-geolocation#quick-start) guide
3) Compare the files in the quick start app `/platforms/android` directory to your app.

Here's a screenshot showing that AdvancedGeolocation is available as a global:
[image]

Here's a screenshot of Android Studio showing the native Android's project directory structure:
[image]

Let me know if you get it going or not?