Esri/cordova-plugin-advanced-geolocation #2: Provide cell id info

andygup, 2016-03-25:
Investigate including cell id info, for example [GsmCellLocation](https://github.com/Esri/cordova-plugin-advanced-geolocation/issues/1#issuecomment-201342751)

roysG, 2016-03-27:
it will be great also to collect the neighbours cells and his info.
But pay attention that in some devices the array of the neighbours return empty array or null and the code need to be change.. :)

andygup, 2016-03-28:
Reference: Android SDK [NeighboringCellInfo](http://developer.android.com/reference/android/telephony/NeighboringCellInfo.html)

andygup, 2016-04-21:
A note from the Android SDK docs related to NeighboringCellInfo: `This method was deprecated in API level 23. Use (@link getAllCellInfo} which returns a superset of the information from NeighboringCellInfo.`

PR #5 (2016-04-22): v0.3.0
No breaking changes.

**Enhancements**
* Closes #2 - Added ability to access cell tower meta-data
* Various doc improvements
commit 057228f4 (2016-04-22): v0.3.0 closes #2
closed 2016-04-22 by andygup