Esri/cordova-plugin-advanced-geolocation #13: if gps is not enabled it also doen't provid network based location

siddhesh-tamhanekar, 2016-05-28:


andygup, 2016-05-31:
Thanks, I'll check it out.

andygup, 2016-05-31:
@siddhesh-tamhanekar I wasn't able to reproduce the behavior on a Google Nexus running 6.0.1. I'm wondering if it's specific to a particular phone manufacturer.

Please let me know about each of the following questions:
- Are you using the latest version of this plugin?
- Device make/model?
- Device operating system version?
- Did you see any warnings or errors in AndroidStudio logcat?
- Are you using the `sample-map.html` sample? Or using some other code?

siddhesh-tamhanekar, 2016-06-01:
Hi andy, Sorry for late reply.

- Did you see any warnings or errors in AndroidStudio logcat?
I am not using androidStudio. just simple cordova cli.

- Are  you using the sample-map.html sample? Or using some other code?
Yes. I have followed instruction given in the 
  'https://www.npmjs.com/package/cordova-plugin-advanced-geolocation' and installed plugin from cordova.

- Device

 model : SM-G313HU

- Android

 Version :4.4.2(Kitkat)

andygup, 2016-06-01:
Gotcha. Please note, as stated in the README this library will only work with Android 5.0 or greater. This is an Android SDK constraint and it may be related to some of your problems. 

AndroidStudio and `adb logcat` are the best way to successfully debug Android operating system errors. You can also try using Chrome remote debugging to see if there are any JavaScript errors showing up on the WebView console. 

Depending on what you find, it may or may not be fixable. I don't have a 4.x Android device so I won't be able to help much other than if you can see any logcat or console debug errors.

andygup, 2016-06-06:
I'm going to go ahead and close this. Feel free to post new questions if they come up.

closed 2016-06-06 by andygup

siddhesh-tamhanekar, 2016-06-07:
Okay.

anup5708, 2017-02-23:
**Still facing this issue**

Just now I added this plugin to my Ionic 2 application and waited for 5 minutes to get network based location but didn't get any update. when I enable GPS it provides me both data i.e., netwok & GPS.

andygup, 2017-02-23:
@anup5708 can you clarify, are you setting the `providers` configuration property to `network` and not getting anything back?

Also what model phone and version of Android are you using?

Reference: [API Doc - Configuration Options](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/api_reference.md#configuration-options-required)

andygup, 2017-02-24:
UPDATE for Android 5.x and 6.x. On these versions of the operating system, if you turn off device `Location` that disables all location providers for security reasons. I'm not aware of any device settings that let you disable GPS and leave other location providers active.

You also have to pay attention to `Mode`. There are three settings:

* **High accuracy**
High accuracy mode uses GPS, Wi-Fi, mobile networks, and other sensors to get the highest-accuracy location. This mode uses Google's Location service to help estimate your device's location faster and more accurately.

* **Battery saving**
Battery saving mode uses sources that use less battery, like Wi-Fi and mobile networks. This mode uses Google's Location service to help estimate your device's location faster and more accurately.

* **Device only**
Device only mode uses only GPS. It doesn’t use Google's Location service to provide location information. This mode may estimate your device's location slower and use more battery.