Esri/cordova-plugin-advanced-geolocation #14: GnssMeasurement

andygup, 2016-07-05:
Opening this issue as a placeholder for the current Android N developments related to GNSS.

References:

Android SDK - API Level 24 [GnssMeasurement Class](https://developer.android.com/reference/android/location/GnssMeasurement.html)

GPS World June 2016 - [Google opens up GNSS pseudoranges](http://gpsworld.com/google-opens-up-gnss-pseudoranges/)

Rokubun Blog, June 30th 2016 - [First look at Android N GNSS raw measurements](http://rokubun.cat/2016/06/30/android-n-preview-gnss-measurements/)

andygup, 2016-07-13:
Update, GpsStatus is deprecated as of API 24. Use GnnsStatus instead. Reference: https://developer.android.com/reference/android/location/GpsStatus.html

mprins, 2017-09-15:
Does this issue mean there is an intent to upgrade to a higher (24+) api level? and break backward compatibility of this plugin?

These classes are still present in API 26 and I suspect will be around for a long time to come.
The android docs in eg. the [`<uses-sdk>`](https://developer.android.com/guide/topics/manifest/uses-sdk-element.html) state that applications are generally forward-compatible wrt to the platform api (or quoting from that linked page  "By design, new versions of the platform are fully backward-compatible.")

andygup, 2017-09-15:
Thanks for asking...I'll clarify this issue by saying I'm leaving the plugin as-is for now. I have no immediate plans to updated to 24+. 

At the time I opened this, there were questions about handling to devices that implement new GNSS higher-resolution capabilities.

andygup, 2017-09-22:
Here's a promising update on the next generation smartphone GPS chips. Broadcom announced that starting in 2018, 'some' phones will have the built-in capability for 30-centimeter accuracy and greater battery savings. Read more  about [here](https://spectrum.ieee.org/tech-talk/semiconductors/design/superaccurate-gps-chips-coming-to-smartphones-in-2018).

obbe79, 2017-10-10:
It would be nice to have the possibility to use the new methods in the GNSSMeasurement class (and in general all of the classes in android.location API). Working with pseudoranges opens up a whole new world! Please upgrade the plug-in! :)

andygup, 2018-05-15:
Update: there is new info about phones having the high-accuracy Broadcom chipset will be available later this year. Here's the [article](https://www.nextbigfuture.com/2018/04/high-end-smartphones-later-in-2018-should-use-broadcom-chip-enables-centimeter-accurate-gps.html) that mentions the Samsung Galaxy Note 9 and the Google Pixel 3 as being potential recipients.

andygup, 2018-07-18:
This was passed along to me, it's an article on dual frequency GNSS support for Android: https://medium.com/@sjbarbeau/dual-frequency-gnss-on-android-devices-152b8826e1c

andygup, 2018-09-21:
Phones running the high accuracy Broadcom BCM47755 chip are starting to become available. 

Here's one example: https://www.mi.com/en/mi8/ and https://www.gsa.europa.eu/newsroom/news/world-s-first-dual-frequency-gnss-smartphone-hits-market