Esri/cordova-plugin-advanced-geolocation #18: How To use in ionic ?

codot91, 2016-07-20:
how to use this in ionic ? i try and get eror  advancedgeolocation undefined

andygup, 2016-07-20:
Make sure the plugin is installed 

The recommended pattern for Cordova plugins is to wait until the platform ready event to fire and then initialize the plugin. Example:

```javascript

someModule.run(function($ionic) {
    // Listen for the ready event
    $ionic.ready(function() {
         if (window.cordova) {
             AdvancedGeolocation.start(function(success){
                 //TO-DO
             }
         }
    });
  });

```

If `.ready` never seems to fire, then Cordova may be in an inconsistent state. Try removing the plugin and then reinstalling it.

```javascript

cordova plugin rm cordova-advanced-geolocation
cordova plugin add cordova-advanced-geolocation

```

andygup, 2016-08-24:
@codot91 I'm going to close this out. If anything else comes up feel free to open a new issue.

closed 2016-08-24 by andygup