Esri/cordova-plugin-advanced-geolocation #22: Fix accidental thread interrupt clearing

andygup, 2016-08-24:
'sendCallback' method in GPS, Cell and Network controllers uses 'Thread.interrupted()' instead of 'Thread.currentThread().isInterrupted()'. Fixing this will remove the possibility of accidentally clearing an interrupt request.

Reference: https://developer.android.com/reference/java/lang/Thread.html#isInterrupted()

closed 2016-08-24 by andygup

andygup, 2018-09-20:
What version of android? Check logcat and let me know if you are getting errors?

aacassandra, 2018-09-25:
this is from chrome console:
`Error JSON: "{\"provider\":\"gps\",\"error\":\"120\",\"msg\":\"GPS location requested but GPS is not available. Check system Location settings\"}"`

`VM198:407 Error no.: 120, Message: GPS location requested but GPS is not available. Check system Location settings, Provider: gps`

sorry for my bad english and for my late response.

From my point of view, I see a repetitive function, when I use the "AdvancedGeolocation.start" function, so I assume that when I turn off the GPS, I will get the function running normally again when I turn on the GPS again. but it was all beyond my expectations, and I really got an error message and the plugin didn't run again after I reactivated my GPS.

I don't know if I miss the code that I don't know or what, maybe you have a solution for this?

while my solution for now is to combine "{function (error) {" from your plugin with HTML5 Web Worker to detect when clients restart their GPS. with the help of the "cordova-plugin-request-location-accuracy" plugin. at that time I repeated the function of your plugin

andygup, 2018-09-25:
@aacassandra What version of Android? Can you send a code sample of what you are trying to do?

aacassandra, 2018-09-25:
im using cordova android@6.4.0 and my device is android 7.1.1

```
    AdvancedGeolocation.start(function(data){
        try{
            var jsonObject = JSON.parse(data);
            switch(jsonObject.provider){
                case "gps":
                    if(jsonObject.latitude != "0.0"){
                        console.log("GPS location detected - lat:"+jsonObject.latitude+", lon: "+jsonObject.longitude+", accuracy: "+jsonObject.accuracy);
                        sessionStorage.setItem('current-loc-lat',jsonObject.latitude);
                        sessionStorage.setItem('current-loc-lon',jsonObject.longitude);
                    }
                break;
            }
        }
        catch(exc){
            console.log("Invalid JSON: " + exc);
        }
    },
    function(error){
        console.log("Error JSON: " + JSON.stringify(error));
        var e = JSON.parse(error);
        console.log("Error no.: " + e.error + ", Message: " + e.msg + ", Provider: " + e.provider);
    },
    {
      "minTime":5000,        // Min time interval between updates (ms)
      "minDistance":1,       // Min distance between updates (meters)
      "noWarn":true,         // Native location provider warnings
      "providers":"gps",     // Return GPS, NETWORK and CELL locations
      "useCache":true,       // Return GPS and NETWORK cached locations
      "satelliteData":false, // Return of GPS satellite info
      "buffer":false,        // Buffer location data
      "bufferSize":0,        // Max elements in buffer
      "signalStrength":false // Return cell signal strength data
    });
```

andygup, 2018-09-27:
The good news is what you are seeing is not a thread exception error. There are several other possibilities:
- Verify that this isn't a browser issue. Try using the `basicgps` sample app here and see if you can reproduce the issue: https://github.com/Esri/quickstart-map-phonegap. 
- Install Android Studio and run logcat to see if there are any native code issues. Native code errors will not be displayed in the browser console.

commit 981efeef (2016-08-24): Closes #22