Esri/cordova-plugin-advanced-geolocation #36: Add new error callbacks to CellLocationProvider

andygup, 2016-09-15:
Don't start if thread is interrupted: https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/controllers/CellLocationController.java#L135

Unable to provide data due to version restriction: https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/controllers/CellLocationController.java#L175

RameshkrishnanV, 2018-04-16:
i'm also facing this issue, my app is crashing only in android 8.... do u know how to fix this issue @andygup

andygup, 2018-04-16:
@RameshkrishnanV can you provide a logcat dump of the error?

RameshkrishnanV, 2018-04-17:
@andygup sorry for the late reply. this is my error log

Fatal Exception: java.lang.RuntimeException: Unable to pause activity {com.mywired.app/com.mywired.app.MainActivity}: java.lang.NullPointerException: Attempt to read from field 'java.lang.Integer android.telephony.PhoneStateListener.mSubId' on a null object reference
       at android.app.ActivityThread.performPauseActivityIfNeeded(ActivityThread.java:3912)
       at android.app.ActivityThread.performPauseActivity(ActivityThread.java:3878)
       at android.app.ActivityThread.performPauseActivity(ActivityThread.java:3852)
       at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:3826)
       at android.app.ActivityThread.-wrap15(Unknown Source)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1609)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6501)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)

andygup, 2018-04-18:
@RameshkrishnanV  

There are a number of potential causes:
- If the `index.html` is calling the `AdvancedGeolocation.kill()` this can happen 
- Or, if `MainActivity` is directly calling `onDestroy()` 
- Or, for some reason some other plugin shutdown Cordova first before this plugin could shutdown.
- Or, something else is using the `PhoneStateListener` Class...the error is very strange because this plugin doesn't use the property `mSubId`, reference: https://android.googlesource.com/platform/frameworks/base/+/master/telephony/java/android/telephony/PhoneStateListener.java#306

andygup, 2018-04-30:
@RameshkrishnanV you might try the suggestion in this issue: https://github.com/Esri/cordova-plugin-advanced-geolocation/issues/61. It took a little while to figure the issue out.

dorianrod, 2018-05-03:
Hi, thanks a lot @andygup. I had the same issue (nexus 5X, android 8.1) and #61 seems ok to resolve this issue. 
Any idea of the impact of commenting the suggested lines? Do you think it could lead to a battery that would discharge faster ?

andygup, 2018-05-03:
@dorianrod I leaning towards saying this will not affect battery life since the `TelephonyManager` seems to have shut down. IMHO it look like an Android bug where if you set `PhoneStateListener.LISTEN_NONE` on any listener then `TelephonyManager` fully shuts down.