Esri/cordova-plugin-advanced-geolocation #39: Errors when targeting Android device

scarabdesign, 2016-10-13:
The plugin works great on the emulator, but targeting my Samsung Galaxy S5 (Android 6.0.1), I'm getting error for all 3 location methods. I'm not sure what to check here:

`V/ActivityThread: updateVisibility : ActivityRecord{8358ffc token=android.os.BinderProxy@a1b3b27 {com.esri.geo/com.esri.geo.MainActivity}} show : true
D/GeolocationPlugin: GEO PERMISSIONS GRANTED.
D/GeolocationPlugin: onResume
D/GeolocationPlugin: validatePermissions()
D/GeolocationPlugin: Counter: 0, display rationale? false
D/GeolocationPlugin: rationale 4: application startup - no perms are set yet
I/GeolocationPlugin: Available location providers: [passive, gps, network]
W/GeolocationPlugin: GPS location requested but GPS is not available. Check system Location settings
W/GeolocationPlugin: Network location requested but the provider is not available. Check system Location settings
I/chromium: [INFO:CONSOLE(174)] "Error JSON: "{\"provider\":\"gps\",\"error\":\"120\",\"msg\":\"GPS location requested but GPS is not available. Check system Location settings\"}"", source: file:///android_asset/www/js/map.js (174)
I/chromium: [INFO:CONSOLE(176)] "Error no.: 120, Message: GPS location requested but GPS is not available. Check system Location settings, Provider: gps", source: file:///android_asset/www/js/map.js (176)
I/chromium: [INFO:CONSOLE(174)] "Error JSON: "{\"provider\":\"network\",\"error\":\"140\",\"msg\":\"Network location requested but the provider is not available. Check system Location settings\"}"", source: file:///android_asset/www/js/map.js (174)
I/chromium: [INFO:CONSOLE(176)] "Error no.: 140, Message: Network location requested but the provider is not available. Check system Location settings, Provider: network", source: file:///android_asset/www/js/map.js (176)
I/Timeline: Timeline: Activity_idle id: android.os.BinderProxy@a1b3b27 time:459102535
E/GeolocationPlugin: Failing gracefully after detecting an uncaught exception on CellLocationController thread. Invalid coordiante value:2147483647
D/GeolocationPlugin: Stopping cell location listeners
I/chromium: [INFO:CONSOLE(174)] "Error JSON: "{\"provider\":\"cell\",\"error\":\"122\",\"msg\":\"Uncaught thread exception. The app may be partial and incorrectly operating. See logcat for full exception dump\"}"", source: file:///android_asset/www/js/map.js (174)
I/chromium: [INFO:CONSOLE(176)] "Error no.: 122, Message: Uncaught thread exception. The app may be partial and incorrectly operating. See logcat for full exception dump, Provider: cell", source: file:///android_asset/www/js/map.js (176)
`

scarabdesign, 2016-10-13:
Opps, sorry I made assumptions about how the permissions are being checked. Deleting issue

andygup, 2016-10-14:
@scarabdesign thanks, glad you got it. 

I've gone ahead and re-opened this issue. When I looked thru the error dump I saw a few issues. The library shouldn't have thrown an uncaught exception on the `CellLocationController` related to `Invalid coordiante value:2147483647`. There's also a minor misspelling in the error message.

I'll take a closer look at that. 

And, let me know if there are any improvements I can make to the documentation to make the security permissions check more clear?

scarabdesign, 2016-10-14:
I was assuming that if location permission was granted it would check if the device also had location services enabled before it attempted to get location data, but it gave errors (codes 120, 122 and 140). However, I can detect those and prompt the user to check if location services is enabled on the device. 

Is it possible to detect the device-wide setting and ask the user if they want to turn it on, then upon confirmation, brings them to the global location services settings? This is how Google Maps does it, but I realize they may be using a private API for that...

andygup, 2016-10-15:
>> I was assuming that if location permission was granted it would check if the device also had location services enabled before it attempted to get location data

Thanks for the feedback and questions. This is one area where the docs could use improvement. The library does check for application, manifest and system/device-level permissions.

Errors 120 and 140 are providing a JavaScript error callback indicating the user has deactivated the GPS and Network Location at the device level. You can use these errors to provide feedback to the user within the HTML app. 

Error 122 is saying there was an internal, fatal error somewhere related to `CellLocationController` and a MAX_INTEGER value wasn't processed correctly. I may need to rethink some of the error handling to help provide more useful stacktrace information when these boundary cases show up. Unfortunately, I haven't been able to reproduce this to fix it in the code. 

>> Is it possible to detect the device-wide setting and ask the user if they want to turn it on, then upon confirmation, brings them to the global location services settings?

Yes. However, this presents a challenging end user interface workflow. You are dealing with multiple system-level settings that exist in different places. For your scenario above, the end user could first be redirected to Location Settings view, and then they have to re-launch the app and choose from a list of options to select [Mobile Network view](http://stackoverflow.com/questions/34893650/redirect-user-to-turn-on-internet-settings) (or whatever its equivalent is on that device). And, this is only one scenario that has to be handled.

One additional challenge, in Android 6 there isn't a way to take the user directly to the application's permissions view. 

I'm open to feedback on this. There aren't any perfect, one-size-fits-all answers on how to handle these scenarios.

This is a lot of info, let me know if you are still stumped?

closed 2016-11-22 by andygup