Esri/cordova-plugin-advanced-geolocation #43: Angular2 / Ionic2

meeximum, 2016-12-08:
Hi, I'm very happy about your plugin!!!!
But I didn't get it working within Ionic2 :-/

andygup, 2016-12-08:
@meeximum please provide info on the following:
- Can you clarify what you tried to do? Provide code if possible, or psuedo-code.
- Describe what didn't work as expected?
- Were there any errors and if so can you copy and paste them here?
- Phone operating system and version?

Thanks!

meeximum, 2016-12-08:
Hi,

when I run `ionic run android --debug `
I get

> Error: cmd: Command failed with exit code 1 Error output:
> E:\git\woodcat-mobile-ionic\platforms\android\src\com\esri\cordova\geolocation\AdvancedGeolocation.java:38: error: cannot find symbol
> import com.esri.cordova.geolocation.model.StopLocation;
>                                          ^
>   symbol:   class StopLocation
>   location: package com.esri.cordova.geolocation.model

meeximum, 2016-12-08:
when I run `cordova plugin list`
I get

> cordova-plugin-advanced-geolocation 1.1.0 "Cordova Advanced Geolocation Plugin - Android"
> cordova-plugin-camera 2.3.0 "Camera"
> cordova-plugin-compat 1.1.0 "Compat"
> cordova-plugin-console 1.0.4 "Console"
> cordova-plugin-device 1.1.3 "Device"
> cordova-plugin-geolocation 2.4.0 "Geolocation"
> cordova-plugin-network-information 1.3.0 "Network Information"
> cordova-plugin-splashscreen 4.0.0 "Splashscreen"
> cordova-plugin-sqlite-2 1.0.4 "SQLitePlugin"
> cordova-plugin-statusbar 2.2.0 "StatusBar"
> cordova-plugin-whitelist 1.3.0 "Whitelist"
> ionic-plugin-keyboard 2.2.1 "Keyboard"

andygup, 2016-12-08:
Here are few thoughts, let me know if these work or not?
- Are you are running the latest version of Cordova 6.x?
- Is your Android API level one of these 21, 22 or 23?
- Is the plugin code running after the platform ready event?

```javascript
    this.platform.ready().then(() => {
         AdvancedGeolocation.start(function(success){
             . . .
         }
    });
```

meeximum, 2016-12-08:
Cordova 6.4.0
Android API level 22
Yes, plugin code is running after device ready

...but I do not come that far, because I get the error when tryin to run the code on the device :-/

ionic uses a kind of wrapper for plugins
https://github.com/driftyco/ionic-native

andygup, 2016-12-08:
Can you share the code for the component or service you are trying to build?

meeximum, 2016-12-09:
you can easiliy reproduce
```
npm install ionic
ionic start test
cd test
ionic platform add android
ionic plugin add https://github.com/esri/cordova-plugin-advanced-geolocation.git --save 
ionic run android
```

andygup, 2016-12-09:
Can you clarify you want to be using angular 2? 

I just need to know for certain before I look into this. It looks like you are running Angular 1. To use Angular 2 run the command `ionic start test --v2`. Not the command contains a `--v2` option. 

One way to verify the angular version is go into your project directory and look under `android\assets\www\lib\ionic\js\angular\angular.js`. Look at the header blurb of the file, for example it may look similar to this:

```javascript
    /**
     * @license AngularJS v1.4.3
     * (c) 2010-2015 Google, Inc. http://angularjs.org
     * License: MIT
     */

```

meeximum, 2016-12-09:
yes angular2, the newest version of ionic2 doesn't need the --v2 parameter ;-)

meeximum, 2016-12-09:
you are right! Sorry, I missed the --v2 in the example but result is the same :-(

andygup, 2016-12-09:
No worries...Gotcha, I'll take a look here shortly.

andygup, 2016-12-09:
Here's the full error dump from Android Studio. It's most likely an Ionic problem because the plugin works as designed on stand-alone Cordova.

I'll try to take a look over the next few days, I'm pretty slammed at the moment. Could have something to do with how the library handles Android 6 permissions.

```
11-09 09:44:27.467 3563-3563/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: com.rsupport.mvagent, PID: 3563
                                                 java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "OPENSSL_add_all_algorithms_noconf" referenced by "/data/app/com.rsupport.mvagent-1/lib/arm/libnet10neon.so"...
                                                     at java.lang.Runtime.loadLibrary(Runtime.java:372)
                                                     at java.lang.System.loadLibrary(System.java:1076)
                                                     at com.rsupport.util.Net10.<clinit>(Net10.java:81)
                                                     at alb.<init>(Net10P2PServer.java:44)
                                                     at com.rsupport.mvagent.connector.ConnectorEngineNet10.hv(ConnectorEngineNet10.java:118)
                                                     at bkk.gF(WebViewerService.java:247)
                                                     at apz.run(Controller.java:468)
                                                     at android.os.Handler.handleCallback(Handler.java:739)
                                                     at android.os.Handler.dispatchMessage(Handler.java:95)
                                                     at android.os.Looper.loop(Looper.java:148)
                                                     at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
11-09 09:44:35.983 5429-5429/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: com.rsupport.mvagent, PID: 5429
                                                 java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "OPENSSL_add_all_algorithms_noconf" referenced by "/data/app/com.rsupport.mvagent-1/lib/arm/libnet10neon.so"...
                                                     at java.lang.Runtime.loadLibrary(Runtime.java:372)
                                                     at java.lang.System.loadLibrary(System.java:1076)
                                                     at com.rsupport.util.Net10.<clinit>(Net10.java:81)
                                                     at alb.<init>(Net10P2PServer.java:44)
                                                     at com.rsupport.mvagent.connector.ConnectorEngineNet10.hv(ConnectorEngineNet10.java:118)
                                                     at bkk.gF(WebViewerService.java:247)
                                                     at apz.run(Controller.java:468)
                                                     at android.os.Handler.handleCallback(Handler.java:739)
                                                     at android.os.Handler.dispatchMessage(Handler.java:95)
                                                     at android.os.Looper.loop(Looper.java:148)
                                                     at android.app.ActivityThread.main(ActivityThread.java:5417)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
11-09 09:44:43.631 6156-6156/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                 Process: com.rsupport.mvagent, PID: 6156
                                                 java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "OPENSSL_add_all_algorithms_noconf" referenced by "/data/app/com.rsupport.mvagent-1/lib/arm/libnet10neon.so"...
                                                     at java.lang.Runtime.loadLibrary(Runtime.java:372)
                                                     at java.lang.System.loadLibrary(System.java:1076)
                                                     at com.rsupport.util.Net10.<clinit>(Net10.java:81)
                                                     at alb.<init>(Net10P2PServer.java:44)
                                                     at com.rsupport.mvagent.connector.ConnectorEngineNet10.hv(ConnectorEngineNet10.java:118)
                                                     at bkk.gF(WebViewerService.java:247)
                                                     at apz.run(Controller.java:468)
                                                     at android.os.Handler.handleCallback(Handler.java:739)
                                                     at android.os.Handler.dispatchMessage(Handler.java:95)
[... cut ...]
                                                       at java.lang.reflect.Method.invoke(Native Method) 
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726) 
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616) 
                                                   
                                                   
                                                   --------- beginning of system

```

andygup, 2016-12-09:
@meeximum okay, I think I figured it out while taking a coffee break. The `Plugin.xml` file is missing a reference to copy `/model/StopLocation.java` over into the project directory. I replaced the missing file and, tah dah, my app started working. 

I'll get that fixed in short order and push a new release. 

In the meantime, you can get running again by creating a `StopLocation.java` file under `com/esri/cordova/geolocation/model/`and using the contents from [here](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/model/StopLocation.java).

closed 2016-12-09 by andygup

meeximum, 2016-12-09:
Wow, great, thx for your quick help and solution!!!!
I will try it with a new example :-)

meeximum, 2016-12-10:
Yes, everything is fine :-) 
Just another question, what is the best approach for getting just one dataset (coordinates)? 
Can I call stop() inside the start function?
I want to implement this behind a function that gives me the actual position back?

best regards
Meex

andygup, 2016-12-11:
Awesome, glad it's working again. 

Yes, you can call `stop()` inside `start()`. Not sure what your requirements are, but retrieving a single location is very inaccurate. As a minimal approach, I always recommend retrieving coordinates until the accuracy values fall within acceptable parameters. Reference: [http://www.andygup.net/html5-geolocation-api-getting-the-best-single-location/](http://www.andygup.net/html5-geolocation-api-getting-the-best-single-location/)

meeximum, 2016-12-25:
Hi Andy,

now as I'm tryin to implement your suggestion to watch the accuracy values, I get stuck :-/
When using Angular2 / Ionic2 environment, I got problmes to reference to variables or events from inside your start function?

```
@Injectable()
export class GeolocationAdvanced {

  public static EVENT_LOCATION_CHANGED = 'location:changed';

  actualLocation: LocationAdvanced;

  constructor(
    private http: Http,
    private platform: Platform,
    private events: Events,
    private connectivityService: Connectivity
  ) {
    this.events.subscribe(GeolocationAdvanced.EVENT_LOCATION_CHANGED, location => {
      this.actualLocation = Array.isArray(location) ? location[0] : location;
    })
  }

  getGeoposition(): Promise<LocationAdvanced> {
    if (this.connectivityService.onDevice) {
      return Promise.resolve(this.actualLocation);
      //return this.locateAdvanced();
    } else {
      return this.locateDefault();
    }
  }

  start() {
    AdvancedGeolocation.start(function (data) {
      try {
        let location = JSON.parse(data);
        console.log(location);
        let actualLocation = new LocationAdvanced(location.longitude, location.latitude, location.accuracy);
        this.events.events.publish(GeolocationAdvanced.EVENT_LOCATION_CHANGED, actualLocation);
        // but this == null :-/ also this.actualLocation doesn't work ...

Maybe you have another trick :-)
}
      catch (error) {
        console.error(error);
      }
    },
      function (error) {
        console.error(error);
      }, {
        "minTime": 500,         // Min time interval between updates (ms)
        "minDistance": 1,       // Min distance between updates (meters)
        "noWarn": true,         // Native location provider warnings
        "providers": "all",     // Return GPS, NETWORK and CELL locations
        "useCache": true,       // Return GPS and NETWORK cached locations
        "satelliteData": false, // Return of GPS satellite info
        "buffer": false,        // Buffer location data
        "bufferSize": 0,       // Max elements in buffer
        "signalStrength": false // Return cell signal strength data
      });
  }

  stop() {
    AdvancedGeolocation.stop();
  }

```

andygup, 2017-01-09:
@meeximum sorry for the delay. Any luck with this? Was `start()` returning any data?

meeximum, 2017-01-09:
no problem, no success on this, I still face the same problem of scoping :-/ start() works, but it's not possible to get any values back to the this obj.

andygup, 2017-01-10:
Gotcha. I'll take a look.

meeximum, 2017-01-10:
After further investigation I found a possible solution :-)
I need to bind this explicitly!

```
private locateAdvanced(): Promise<LocationAdvanced> {
    //this.platform.ready().then(() => {
    return new Promise(function (resolve, reject) {

      AdvancedGeolocation.start(function (data) {
        try {
          let location = JSON.parse(data);

          AdvancedGeolocation.stop();
          resolve(new LocationAdvanced(location.longitude, location.latitude, location.accuracy));
        }
        catch (error) {
          // Fallback
          return this.locateDefault()
            .catch(error => {
              reject(error);
            });
        }
      }.bind(this), // bind explicitly to this
        function (error) {
          // Fallback
          return this.locateDefault()
            .catch(error => {
              reject(error);
            });
        }, {
          "minTime": 500,         // Min time interval between updates (ms)
          "minDistance": 1,       // Min distance between updates (meters)
          "noWarn": true,         // Native location provider warnings
          "providers": "all",     // Return GPS, NETWORK and CELL locations
          "useCache": true,       // Return GPS and NETWORK cached locations
          "satelliteData": false, // Return of GPS satellite info
          "buffer": false,        // Buffer location data
          "bufferSize": 0,       // Max elements in buffer
          "signalStrength": false // Return cell signal strength data
        });
    }.bind(this)); // bind explicitly to this
    //});
  }
```

andygup, 2017-01-10:
Awesome! I haven't been able to test. The latest Ionic build has some fatal errors on MacOS that prevent it from installing the Android platform.

commit 17ee5506 (2016-12-09): Closes #43