Esri/cordova-plugin-advanced-geolocation #49: Blocking other native plugins

paidTrainee27, 2017-07-09:
Hi, 
Thanks alot to you for building such a plugin. 
Currently I'm Using it along with Ionic 2/3 framework for hybrid application built on cordova and angular
My problem is that the moment I call for start method ....I do get the co-ordinates but other plugins stop working like or example Sqlite plugin of cordova.
Like I 'm using your plugin and sqlite plugin on device ready, but after start method my sqlite plugin stops responding ...like no error and no success response
Ihave tried all possible alternative like trycatch,returning promise using window approach to call sqliteplugin etc..

andygup, 2017-07-10:
Hi, I have two suggestions to try and figure out what's going on the easiest way possible.

First, I recommend trying remote debugging the application, see [this article](https://developers.google.com/web/tools/chrome-devtools/remote-debugging/) for more details. 

Second suggestion if you can't get anything useful from remote debugging is to install Android Studio and use the Android Monitor to view logcat messages. Logcat dumps all your phone's operating system messages to a console: https://developer.android.com/studio/profile/android-monitor.html. 

Once you are able to capture error messages please post them here for review. It's very rare that plugins fail without throwing some sort of error message.

andygup, 2017-07-31:
@paidTrainee27 I'm closing this for now. If you have additional information or questions feel free to reopen.

closed 2017-07-31 by andygup

ankitagarwal-ttl, 2018-04-26:
Hi,
I am getting same issue:

Please find the logs:

 E/SQLitePlugin: couldn't add to queue
                                                 java.lang.InterruptedException
                                                     at java.util.concurrent.locks.AbstractQueuedSynchronizer.acquireInterruptibly(AbstractQueuedSynchronizer.java:1218)
                                                     at java.util.concurrent.locks.ReentrantLock.lockInterruptibly(ReentrantLock.java:312)
                                                     at java.util.concurrent.LinkedBlockingQueue.put(LinkedBlockingQueue.java:310)
                                                     at io.sqlc.SQLitePlugin.executeAndPossiblyThrow(SQLitePlugin.java:159)
                                                     at io.sqlc.SQLitePlugin.execute(SQLitePlugin.java:89)
                                                     at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
                                                     at org.apache.cordova.PluginManager.exec(PluginManager.java:132)
                                                     at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:57)
                                                     at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
                                                     at org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
                                                     at org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:9)
                                                     at android.os.Handler.dispatchMessage(Handler.java:102)
                                                     at android.os.Looper.loop(Looper.java:154)
                                                     at android.os.HandlerThread.run(HandlerThread.java:61)

ankitagarwal-ttl, 2018-04-26:
Native Sqlite plugin stops working after using advance geo location.

Please help me to solve this issue.

andygup, 2018-04-26:
@ankitagarwal-ttl Thanks for the call stack info. There isn't an easy fix for this issue. 

The error message above shows an issue with `ConcurrentHashMap` in SQLitePlugin.java. This type of map is synchronized and allows for thread-safe read concurrency. But, apparently something is going wrong and the error isn't being handled correctly. At a glance I don't see any implementation of `Runnable` in that .java file. It's possibly a conflict with the Cordova `ExecutorService`.

A second issue is Cordova is not correctly checking for `Thread.currentThread().isInterrupted()`. These items in combination would cause the plugin to stop working. 

I'm not sure why or how this geolocation plugin could be stepping on SQLitePlugin to cause an issue like that. 

Potential workarounds for testing why this is happening:
* [Easiest] You could try adding an interrupt check and exception handling to CordovaBridge.java at line 57 in your project. If we can get more information on where interrupt is occurring that would help isolate the issue, or potentially keep Cordova from breaking.
* [Much more work] Do some customization of the SQLitePlugin.java by using Cordova threadpool and implement runnables around the ConcurrentHashMap methods.
* Or I've completely read this wrong and missed something simple?

Reference:
* In CordovaBridge, the code does not check for an interrupt before asking for the current thread, and they don't implement `setUncaughtExceptionHandler()`: [CordovaBridge.java:57](https://github.com/apache/cordova-android/blob/master/framework/src/org/apache/cordova/CordovaBridge.java#L57)

* [CordovaPlugin.java:98](https://github.com/apache/cordova-android/blob/master/framework/src/org/apache/cordova/CordovaPlugin.java#L96)

* [SQLitePLugin.java:146](https://github.com/litehelpers/Cordova-sqlite-storage/blob/storage-master/src/android/io/sqlc/SQLitePlugin.java#L146)

junbonglee, 2018-10-26:
Did anyone find solution to this problem? We can't use cordova-advanced-geolocation and ionic native storage at same time. Stuck in the middle of nowhere.

junbonglee, 2018-10-30:
I'm having same issue when trying to use cordova-plugin-advanced-geolocation and ionic-storage(if it tryse to use sqlite-driver). Is there any update on this error? I'd really like to use advanced-geolocation but it's even harder to give up sqlite.