Esri/cordova-plugin-advanced-geolocation #52: Neighboring Cell Info

Marcophono2, 2017-09-04:
Hello!

I am very happy with this plugin. It is MUCH faster than the standard geolocation plugin. (by the way it also works on OS 7). The only thing I am missing are informations for the neighbor cell. Is there a trick to add this data too?

Best regards
Marc

Marcophono2, 2017-09-04:
Sorry. I was wrong. The cell infos are not coming in in one array. They come in one by one. In my test app I write the caught data into a div field and didn't realize that the later coming info overwrote the existing ones. It was too fast for my eyes. ;-) Just collecting the data was an easy solution. 
Thank you for this wonderful working plugin! As far as I ca say the only one available working so fast and returning (also) all cell data.

Best regards
Marc

andygup, 2017-09-05:
Glad the plugin helped! Sounds like your question was answered and we close this out?

For others looking for the same info - neighboring cells info is gathered via the Android SDK [TelephonyManager.getAllCellInfo()](https://developer.android.com/reference/android/telephony/TelephonyManager.html#getAllCellInfo()) method. You can find that code in this plugin in the [CellLocationController.java](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/controllers/CellLocationController.java#L165) file.

Marcophono2, 2017-09-05:
Yes, you can close it, thank you! Aside that I receive 2147483647 (32bit max-length) for all values of the neighboring cell. But that seems to be a problem deeper in the TelephonyManager.getAllCellInfo(). I also read that this senseless value points to invalid values sent by the cell itsself but I do not believe that. I found an Android app which was able to locate the neighboring cell correctly while my app only displays 2147483647. I think the other app developer added a workaround to the Android code. The neighboring cell's info is not urgent for me. Just wanted to find out what is possible andwhere the limits are. :-)

Best regards
Marc

andygup, 2017-09-05:
>>I found an Android app which was able to locate the neighboring cell correctly while my app only displays 2147483647.

Yep, that value means there is no valid values being returned. Interesting you had another app that was working since those values are returned by the operating system. I wonder if the other app uses the old [getNeighboringCellInfo()](https://developer.android.com/reference/android/telephony/TelephonyManager.html#getNeighboringCellInfo()) method which was deprecated at v23. That's the only work-around I can think of short of creating a non-trivial operating system patch.

closed 2017-09-05 by andygup