Esri/cordova-plugin-advanced-geolocation #53: Problem to get AdvancedGeolocation object Ionic3

mbahojlo, 2017-10-02:
From the begin... I will start project...
`c:\xampp\htdocs>Ionic start ionicIssueGeo`
`c:\xampp\htdocs>cd ionicIssueGeo`
`c:\xampp\htdocs\ionicIssueGeo>ionic --version`
`3.12.0`
`c:\xampp\htdocs\ionicIssueGeo>cordova -version`
`7.0.1`
`c:\xampp\htdocs\ionicIssueGeo>ionic serve`
[image]
Then next test
`c:\xampp\htdocs\ionicIssueGeo>ionic cordova run android`
[image]

Then add plugin:
`c:\xampp\htdocs\ionicIssueGeo>cordova plugin add https://github.com/esri/cordova-plugin-advanced-geolocation.git`
`cordova plugin list`
`cordova-plugin-advanced-geolocation 1.1.0 "Cordova Advanced Geolocation Plugin - Android"`
`cordova-plugin-device 1.1.4 "Device"`
`cordova-plugin-splashscreen 4.0.3 "Splashscreen"`
`cordova-plugin-statusbar 2.2.2 "StatusBar"`
`cordova-plugin-whitelist 1.3.1 "Whitelist"`
`ionic-plugin-keyboard 2.2.1 "Keyboard"`

Then test from quick start guide
Edit config.xml file
[image]
Test app:
[image]
Test on live device
[image]

So far so good...
And problems now begin:
Want to put info from GEO to home tab in ionic template app so Edit home.html templete:
[image]
And home.ts:
[image]

**So the problem is that i'm trying to get AdvancedGeolocation from plugin:**

`declare var cordova; //<===`
`declare var AdvancedGeolocation; `
.....
`  console.log(cordova.plugins);      //EMPTY <===`
`  console.log(AdvancedGeolocation);  //EMPTY <===`
`  console.log((<any>window).AdvancedGeolocation); ;  //EMPTY <===`
`  cordova.plugins.AdvancedGeolocation.start(function(success){ `

**but AdvancedGeolocation object is undefined...
How to get reference to AdvancedGeolocation?**

andygup, 2017-10-02:
@mbahojlo Sounds like a scope issues, and you may need to explicitly bind to `this`. For example: https://github.com/Esri/cordova-plugin-advanced-geolocation/issues/43#issuecomment-271682035. Give that a try and let me know if it works?

andygup, 2017-10-02:
@mbahojlo I ran a quick test using Ionic 3.12 (latest). The following code worked for me. Note I switched to using `OnInit` and declared `AdvancedGeolocation` as `any`.

```javascript

import {Component, OnInit} from '@angular/core';

import { NavController } from 'ionic-angular';

declare var AdvancedGeolocation:any;

@Component({
  selector: 'page-home',
  templateUrl: 'home.html'
})
export class HomePage implements OnInit{

  constructor(public navCtrl: NavController) { }

  ngOnInit() {

    start();

    function start(){

      AdvancedGeolocation.start(function(success){

          try{
            var jsonObject = JSON.parse(success);
            console.log("Provider " + jsonObject);
            switch(jsonObject.provider){
              case "gps":
                //TODO
                break;

              case "network":
                //TODO
                break;

              case "satellite":
                //TODO
                break;

              case "cell_info":
                //TODO
                break;

              case "cell_location":
                //TODO
                break;

              case "signal_strength":
                //TODO
                break;
            }
          }
          catch(exc){
            console.log("Invalid JSON: " + exc);
          }
        },
        function(error){
          console.log("ERROR! " + JSON.stringify(error));
        },
        ////////////////////////////////////////////
        //
        // REQUIRED:
        // These are required Configuration options!
        // See API Reference for additional details.
        //
        ////////////////////////////////////////////
        {
          "minTime":500,         // Min time interval between updates (ms)
          "minDistance":1,       // Min distance between updates (meters)
          "noWarn":true,         // Native location provider warnings
          "providers":"all",     // Return GPS, NETWORK and CELL locations
          "useCache":true,       // Return GPS and NETWORK cached locations
          "satelliteData":false, // Return of GPS satellite info
          "buffer":false,        // Buffer location data
          "bufferSize":0,         // Max elements in buffer
          "signalStrength":false // Return cell signal strength data
        });
    }
  }
}

```

kumarkundan, 2017-10-08:
@andygup  .....it's working bro ...thanks , i was searching of something like this from a long time , may be month or so. thanks a lot again

andygup, 2017-10-09:
Good deal, I'll go ahead and close this issue.

closed 2017-10-09 by andygup