Esri/cordova-plugin-advanced-geolocation #57: Ionic 2 / ionic 3 : unable to get "gps" , "network" and "satellite" ' s provider objects

kumarkundan, 2017-11-10:
This plugin was working fine for me till yesterday , and i was using it in one of my ionic2 apps. But suddenly from today it's stopped working . It's not working for any of my 2-3 ionic apps. ealier it was working for all those apps.

On Debugging and checking console , I was surprise that in success of  ` AdvancedGeolocation.start((success)` i was only getting these two types of provider objects :- 

{"provider":"cell_info","type":"wcdma","timestamp":1510306725281,"cid":2147483647,"lac":2147483647,"mcc":2147483647,"mnc":2147483647,"psc":252}

`{"provider":"cell_location","type":"gsm","timestamp":1510307383757,"cid":167954,"lac":25,"psc":-1}`

i am not getting "gps" , "network" and "satellite" ' s provider objects at all.

andygup, 2017-11-10:
Did you see any native Android errors in [logcat (available via Android Studio)](https://developer.android.com/studio/debug/am-logcat.html) when you started the app? That is always the first thing that I check.

Here are some suggestions:
- Double check that location is enabled for each app
- Run a quick Geolocation test. Open this browser-based test app and see if it returns a location? https://esri.github.io/html5-geolocation-tool-js/

kumarkundan, 2017-11-21:
hii @andygup , today i again checked . now everything is working fine. For your info , i have not changed a single word in my code base. 

Now i am getting all the objects in provider , earlier i was getting only `cell_info` and `cell_location` . That means , i was not getting other provider's objects.

I want to ask you this . Whether `**cordova-plugin-advanced-geolocation**` is reliable or not . I can use this in production environment or not. Because it seems like , sometimes it's working and sometimes it's not.

You tell me , what to do . Again I have not changed a single word in code base and now it's working fine .

B/W thanks a lot for helping me out.

andygup, 2017-11-21:
@kumarkundan Given the intermittent nature of the problem, this appears to be an issue with your cell phone provider. For example, the value `2147483647` in the response object tells me that a particular cell phone tower is simply not providing the information you want. 

As far as I know, there are generally no international requirements that cellular towers provide any of this information consistently or at all.

My recommendation is to write your code to gracefully handle the situations where the cell provider does not provide information.

Does that make sense?

andygup, 2017-12-07:
Closing this issue. It's a problem with the cellular service provider.

closed 2017-12-07 by andygup