Esri/cordova-plugin-advanced-geolocation #58: Error message after close the App.

rochadsouza, 2017-11-20:
Hi,
I'm using Advanced Geolocation and after adding it to my project, when I close the App is displayed an error message saying "[App name] keeps stopping".  It seem a kind of conflict with another plugin because if I turn the Cordova Google Map Plugin off, this issue does not happens.

In ADB logcat I'm having the following errors:

11-20 18:12:00.345  3112  3112 D Layer   : applyEffect ## com.sec.android.app.launcher/com.sec.android.app.launcher.activities.LauncherActivity
11-20 18:12:00.345  3112  3112 D Layer   : applyEffect ## animMode=3, target=2, pixType=1
11-20 18:12:00.345  3112  3112 D Layer   : applyEffect ## pixAnim anim=13, timeMs=0, value=0.00, interp=0
11-20 18:12:00.345  3112  3112 D Layer   : applyEffect ## pixAnim anim=13, timeMs=0, value=132.00, interp=3
**11-20 18:12:00.349  3682  3785 D WindowManager: finishDrawingWindow: Window{f95c19d0 u0 Application Error: com.ionicframework.takeappproducao} mDrawState=DRAW_PENDING**
11-20 18:12:00.349  3682  3785 D ViewRootImpl@8b73763[takeappproducao]: MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
11-20 18:12:00.349  3682  3819 D LockSettingsStorage: mSKTLockState = 2
11-20 18:12:00.352  3682  4201 D WindowManager: adjustSystemUiVisibilityLw : vis= 0x4008
**11-20 18:12:00.354  3682  3785 D WindowManager: finishDrawingWindow: Window{f95c19d0 u0 Application Error: com.ionicframework.takeappproducao} mDrawState=HAS_DRAWN
**
11-20 18:12:00.358  3682  6830 D WindowManager: finishDrawingWindow: Window{a948105d0 u0 com.sec.android.app.launcher/com.sec.android.app.launcher.activities.LauncherActivity}

[image]

andygup, 2017-11-20:
There don't seem to be any errors related to this plugin. I'm not familiar with the Cordova Google Map Plugin. It's possible there are some native Android platform event issues between it and this plugin. 

Are you calling `AdvancedGeolocation.kill()` in your app? 

Can you send a larger adb dump that begins at the [`GeolocationPlugin: onStop`](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/AdvancedGeolocation.java#L436) message?

rochadsouza, 2017-11-21:
Hi @andygup, thank your for your quick answer. 

I'm not calling `AdvancedGeolocation.kill()`, and already tied to call `AdvancedGeolocation.stop()` when the app goes to background mode and `AdvancedGeolocation.start(...)` when it comes form background mode - everything goes right till I close the app and open it again, the app start up is slowly and Android OS asks if I want to "Wait for the App".

As you asked, here is ADB from `GeolocationPlugin: onStop`. Thank you for your attention!

11-21 15:12:43.977 22569 22569 D GeolocationPlugin: onStop
11-21 15:12:43.977 22569 22569 D GeolocationPlugin: Stopping geolocation
11-21 15:12:43.977  3682  4485 D libgps  : OnGpsExtensionMessage: message_id(1), data(0x71a07fd044), size(2728)
11-21 15:12:43.978  3682  4485 D libgps  : proxy__gnss_sv_status_cb: called. num_svs(0)
11-21 15:12:43.983  3682  4792 D LockSettingsStorage: mSKTLockState = 2
11-21 15:12:43.985 22569 22569 I chromium: [INFO:CONSOLE(69)] "app running in background", source: file:///android_asset/www/js/app.js (69)
11-21 15:12:43.986 22569 22569 I chromium: [INFO:CONSOLE(164)] "Go to Background fn", source: file:///android_asset/www/js/controllers/map_controllers.js (164)
11-21 15:12:43.988  3682  4792 D InputDispatcher: Focus entered window: 4083
11-21 15:12:43.996  4083  4083 D RecentsSlidingView: onMeasure
11-21 15:12:43.996  4083  4240 D mali_winsys: EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
11-21 15:12:43.996  4083  4083 D RecentsSlidingView: onLayout(false, 0, 0, 1080, 1920)
11-21 15:12:43.997  4083  4083 D TaskStackView: prepareForEnterAnimation
11-21 15:12:43.998  4083  4083 D ToggleSlider: onReceive : android.intent.action.CLOSE_SYSTEM_DIALOGS
11-21 15:12:43.998  4083  4083 D vol.SecVolumeDialog: dismissH reason: 2
11-21 15:12:43.998  4083  4083 D vol.SecVolumeDialog: dismissH : false
11-21 15:12:43.998  4083  4083 D Tile.MobileDataTile: action:android.intent.action.CLOSE_SYSTEM_DIALOGS
11-21 15:12:44.000  3682  7095 V WindowManager: Relayout Window{92df660d0 u0 com.ionicframework.takeappproducao/com.ionicframework.takeappproducao.MainActivity}: viewVisibility=4 req=1080x1920 WM.LayoutParams{(0,0)(fillxfill) sim=#110 ty=1 fl=#81810100 pfl=0x20000 wanim=0x103038a vsysui=0x600 needsMenuKey=2 naviIconColor=0}
11-21 15:12:44.000  4083  4083 D SwipeDoorsillDetector: reset()com.android.systemui.statusbar.phone.NotificationPanelView.resetViews:807 com.android.systemui.statusbar.phone.PanelBar.collapsePanel:173 com.android.systemui.statusbar.phone.PhoneStatusBar.animateCollapsePanels:3736 com.android.systemui.statusbar.phone.PhoneStatusBar.animateCollapsePanels:3692
11-21 15:12:44.001  3682  7095 D LockSettingsStorage: mSKTLockState = 2
11-21 15:12:44.005  4083  4083 D ToggleSlider: onReceive : android.intent.action.CLOSE_SYSTEM_DIALOGS
11-21 15:12:44.005  4083  4083 D ToggleSlider: onReceive : android.intent.action.CLOSE_SYSTEM_DIALOGS
11-21 15:12:44.005  4083  4083 D ViewRootImpl@afeb898[RecentsActivity]: MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 72 - 0, 0) or=1
11-21 15:12:44.005  4083  4083 D ViewRootImpl@afeb898[RecentsActivity]: MSG_WINDOW_FOCUS_CHANGED 1
11-21 15:12:44.006  3682  7093 D WindowManager: adjustSystemUiVisibilityLw : vis= 0x4708
11-21 15:12:44.006  4083  4083 V InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@dff86b1 nm : com.android.systemui ic=null
11-21 15:12:44.006  4083  4083 I InputMethodManager: [IMM] startInputInner - mService.startInputOrWindowGainedFocus
11-21 15:12:44.007  3682  3749 D InputMethodManagerService: windowGainedFocus mCurrentFocusedUserId - 0 and mSecureKeypadEnabled-false
11-21 15:12:44.007  3682  3749 D InputMethodManagerService: mDesktopModeManager null!
11-21 15:12:44.007  3682  3749 D InputTransport: Input channel constructed: fd=392
11-21 15:12:44.007  3682  3749 D InputTransport: Input channel destroyed: fd=392
11-21 15:12:44.008  4083  4083 D InputTransport: Input channel constructed: fd=139
11-21 15:12:44.009  4714  4725 D ForegroundUtils: could not check pending caller
11-21 15:12:44.009 22569 22569 D InputTransport: Input channel destroyed: fd=130
11-21 15:12:44.013  3682  4791 I LocationManagerService: remove a7aa93e
11-21 15:12:44.013  3682  4791 I LocationManagerService: removeUpdates, receiver.requestedID = a7aa93e
11-21 15:12:44.014  3682  4791 D LocationManagerService: provider request: fused ProviderRequest[ON interval=+30m0s0ms]
11-21 15:12:44.014  3682  5733 D PackageManager: getComponentMetadataForIconTray : com.ionicframework.takeappproducao.MainActivity does not exist in mServices
11-21 15:12:44.014  3682  5733 D PackageManager: getComponentMetadataForIconTray : com.ionicframework.takeappproducao.MainActivity does not exist in mProviders
11-21 15:12:44.014  3682  5733 D PackageManager: getComponentMetadataForIconTray : com.ionicframework.takeappproducao.MainActivity does not exist in mReceivers
11-21 15:12:44.015  4083  4083 I ApplicationPackageManager: load=com.ionicframework.takeappproducao, bg=144-144, dr=144-144, forDefault=true
11-21 15:12:44.016  4480  4480 D FusedRequestManager_FLP: manageLocationRequest, remove request from com.ionicframework.takeappproducao with a7aa93e through LocationManagerService, provider = 1
11-21 15:12:44.017  4083  4083 I ApplicationPackageManager: scaled rate=201.0, size=144, alpha=0, hold=26, target=144
[... cut ...]
11-21 15:12:50.169 22569 22653 E PluginManager: 	at android.os.Handler.dispatchMessage(Handler.java:102)
11-21 15:12:50.169 22569 22653 E PluginManager: 	at android.os.Looper.loop(Looper.java:154)
11-21 15:12:50.169 22569 22653 E PluginManager: 	at android.os.HandlerThread.run(HandlerThread.java:61)
11-21 15:12:50.283  3682  3819 D LockSettingsStorage: mSKTLockState = 2
11-21 15:12:50.415  3682  3682 D CustomFrequencyManagerService: releaseDVFSLockLocked : Getting Lock type frm List : DVFS_MIN_LIMIT  frequency : 728000  uid : 1000  pid : 3682  tag : GESTURE_DETECTED@CPU_MIN@42

andygup, 2017-11-21:
@rochadsouza I see you closed the issue. Did you figure it out?

rochadsouza, 2017-11-21:
Sorry, it was a mistake.

andygup, 2017-11-21:
@rochadsouza no worries. It looks like Google Maps is throwing an error because it called `onDestroy()`. That event is then being captured by the AdvancedGeolocation plugin causing a cascade failure. Anything with a capital `E` in the logcat dump is an error:

```
11-21 15:12:50.169 22569 22653 E PluginManager: Uncaught exception from plugin
11-21 15:12:50.169 22569 22653 E PluginManager: java.util.concurrent.RejectedExecutionException: Task java.util.concurrent.FutureTask@22efcea rejected from java.util.concurrent.ThreadPoolExecutor@aff22bf[Shutting down, pool size = 4, active threads = 4, queued tasks = 0, completed tasks = 53]
11-21 15:12:50.169 22569 22653 E PluginManager: at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2049)

. . .

1-21 15:12:49.597 22569 22569 D GeolocationPlugin: onDestroy
11-21 15:12:49.597 22569 22569 D GeolocationPlugin: Stopping geolocation
11-21 15:12:49.597 22569 22569 D GeolocationPlugin: Attempting to shutdown cordova threadpool

 . . .

11-21 15:12:49.605 22569 22569 E AndroidRuntime: java.lang.RuntimeException: Unable to destroy activity {com.ionicframework.takeappproducao/com.ionicframework.takeappproducao.MainActivity}: java.util.concurrent.RejectedExecutionException: Task java.util.concurrent.FutureTask@60cb0de rejected from java.util.concurrent.ThreadPoolExecutor@aff22bf[Shutting down, pool size = 4, active threads = 4, queued tasks = 0, completed tasks = 53]

 . . .

11-21 15:12:50.169 22569 22653 E PluginManager: at plugin.google.maps.CordovaGoogleMaps.execute(CordovaGoogleMaps.java:279

```

When you try to restart the app the Google plugin is trying to [`execute`](https://github.com/mapsplugin/cordova-plugin-googlemaps/blob/master/src/android/plugin/google/maps/CordovaGoogleMaps.java#L279) a command but the Cordova threadpool has crashed.  

**Recommendations:**
1) In AdvancedGeolocation.java (`/platforms/android/java`) comment out the following lines:
* [Line 135](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/AdvancedGeolocation.java#L135)
* [Line 443 - 448](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/AdvancedGeolocation.java#L443-L448)
2) Then re-build the app, run it and try closing/opening it again.

Also, shameless plug, this plugin works just fine with the [ArcGIS API for JavaScript ](https://developers.arcgis.com/javascript/) mapping code included in this repo. :-)

andygup, 2017-12-07:
I'm going to close this. If you see additional problems feel free to reopen.

closed 2017-12-07 by andygup