Esri/cordova-plugin-advanced-geolocation #59: the plugin freezes my app - ionic 3

alejox21, 2017-12-27:
Hello, I'm using the plugin in my app for get the gps location timestamp (beacuse I need the real datetime from the place where i'm, independent from the device, and also internet network connection may not be available or possible), the value obtained it's ok, but freezes the app...

ionic 3.19.0
cordova 7.0.1
device: android 5.1

my code:
-----------------------------------------------------------------
service.ts
declare var AdvancedGeolocation:any;
.
.
.
//i have used some code from old comments :)
obtenerTiempo():Promise<string>{
    return new Promise(function (resolve, reject) {
      AdvancedGeolocation.start(function (data) {
        try{
          let location = JSON.parse(data);

         // AdvancedGeolocation.stop();
          resolve(location.timestamp);
        }
        catch (error) {
          reject(error);
        }
      }.bind(this), // bind explicitly to this
        function (error) {
          reject(error);
        }, {
          "minTime": 500,         // Min time interval between updates (ms)
          "minDistance": 3,       // Min distance between updates (meters)
          "noWarn": true,         // Native location provider warnings
          "providers": "gps",     // Return GPS, NETWORK and CELL locations
          "useCache": true,       // Return GPS and NETWORK cached locations
          "satelliteData": false, // Return of GPS satellite info
          "buffer": false,        // Buffer location data
          "bufferSize": 0,        // Max elements in buffer
          "signalStrength": false // Return cell signal strength data
        });
    }.bind(this));
  };

//stop() function
  detenerTiempo():Promise<string>{
    return new Promise((resolve, reject) => {
      AdvancedGeolocation.stop(
        function(success){
          var jsonObject = JSON.parse(success);
          resolve(JSON.stringify(jsonObject.stopLocation));
        },
          function(error){
            reject(JSON.stringify(error));
          },
      );
    })
  };
-------------------------
page.ts
tiempo : any = 0;
.
.
.
//function button start service, sets "tiempo" variable
botonTiempo(){
    //obtener tiempo advanced geolocation
    this.authService.obtenerTiempo().then( obtenido =>{
        this.tiempo = obtenido;
      }
    ).catch((err) => {
                     this.handleError(err)
                   }
                 );
  }

//function stop()
  botonTiempoDetener(){
    this.authService.detenerTiempo().then( detenido =>{

        }
      ).catch((err) => {
                       this.handleError(err)
                     }
                   );
  }
-----------------------
page.html
<ion-content padding>
  <p><strong>Timestamp:</strong>{{tiempo}}</p><br/>
  <button ion-button large full (tap)="botonTiempo()" color="danger">
    Tiempo
  </button>
  <br/>
  <button ion-button large full (tap)="botonTiempoDetener()" color="danger">
    Tiempo Detener
  </button>
  <br/>
</ion-content>
----------------------------------------------------------------------

sorry for my bad english... (also im still noob in ionic), thank for your help

alejox21, 2017-12-28:
sorry my bad, hope I edited properly

andygup, 2018-01-09:
You'll need to remote debug your app. Do you see any errors?

Reference: https://developers.google.com/web/tools/chrome-devtools/remote-debugging/

andygup, 2018-02-24:
Closing out. Feel free to reopen.

closed 2018-02-24 by andygup

BossAustralia, 2019-07-18:
Hello,

I have the same problem, the freeze, but I don't receive any error in the console.

JoelDoitlean, 2019-07-18:
Hello,

I have the same problem, the freeze, but I don't receive any error in the console.

alejox21, 2019-07-18:
Hello, sorry, but at the end i changed to another plugin. Looking at my code above now, maybe the attributes configuration was not configured properly and the method stopTime() it was not well written by me, then time does not stop

andygup, 2019-07-19:
If you don't have any browser console errors you will need to debug the phone using logcat and look for device errors coming from the Android operating system. This may or may not be related to incorrect thread pooling design in the Cordova bridge.

I'm not able reproduce and without additional information or error messages I can't figure it out.