Esri/cordova-plugin-advanced-geolocation #60: Location Radious

soganideepak, 2018-03-25:
Hello
i am unable to find how i can set the geo radius to get my nearest location. right now when i enabled location and lanuch the google map using provided lat long (getting from the plugin), the distance approx. 500 meters from my current location. 
But when i run second time it give exact location. 
Please let me know where i do the changes.

Thanks

andygup, 2018-03-26:
What you are seeing is normal location variation from the phone GPS and it is not related to this library. GPS is very inaccurate when it is first activated. The only way to minimize this is to set `buffer` in the configuration options and let the GPS gather multiple locations for some period of time.

andygup, 2018-04-05:
Closing this out...if something else comes up feel free to reopen.

closed 2018-04-05 by andygup