Esri/cordova-plugin-advanced-geolocation #61: NullPointerException on CellLocationController.stopLocation on Android 8.1

Dan890, 2018-04-26:
java.lang.RuntimeException: 
  at android.app.ActivityThread.performPauseActivityIfNeeded (ActivityThread.java:3912)
  at android.app.ActivityThread.performPauseActivity (ActivityThread.java:3878)
  at android.app.ActivityThread.performPauseActivity (ActivityThread.java:3852)
  at android.app.ActivityThread.handlePauseActivity (ActivityThread.java:3826)
  at android.app.ActivityThread.-wrap15 (Unknown Source)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1609)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:164)
  at android.app.ActivityThread.main (ActivityThread.java:6501)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:438)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:807)
Caused by: java.lang.NullPointerException: 
  at android.telephony.TelephonyManager.listen (TelephonyManager.java:3574)
  at com.esri.cordova.geolocation.controllers.CellLocationController.stopLocation (CellLocationController.java:142)
  at com.esri.cordova.geolocation.AdvancedGeolocation.stopLocation (AdvancedGeolocation.java:348)
  at com.esri.cordova.geolocation.AdvancedGeolocation.onPause (AdvancedGeolocation.java:433)
  at org.apache.cordova.PluginManager.onPause (PluginManager.java:211)
  at org.apache.cordova.CordovaWebViewImpl.handlePause (CordovaWebViewImpl.java:431)
  at org.apache.cordova.CordovaActivity.onPause (CordovaActivity.java:245)
  at android.app.Activity.performPause (Activity.java:7170)
  at android.app.Instrumentation.callActivityOnPause (Instrumentation.java:1409)
  at android.app.ActivityThread.performPauseActivityIfNeeded (ActivityThread.java:3901)

andygup, 2018-04-26:
@Dan890 I have some questions:
* Which Android phone & OS version?
* Does it happen consistently?

Seems like it could be a bug in this code here since the listener was already set to none: https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/controllers/CellLocationController.java#L142

Dan890, 2018-04-26:
@andygup
i see it in the google paly logs, 
from what i see it happen consistntly, all the users with Android 8.1 get this exception.

andygup, 2018-04-26:
Can you try commenting out this line, recompile then test?

https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/controllers/CellLocationController.java#L142

Example:

```java
if(_phoneStateListener != null && _telephonyManager != null){
            _telephonyManager.listen(_phoneStateListener, PhoneStateListener.LISTEN_NONE);
            // _telephonyManager.listen(_signalStrengthListener, PhoneStateListener.LISTEN_NONE);
            _phoneStateListener = null;
            _signalStrengthListener = null;
            _telephonyManager = null;
```

Dan890, 2018-04-29:
seems that this fix the problem.
Thanks!!!

andygup, 2018-04-30:
Good deal! I'll leave this open as an enhancement. I suspect there were some tweaks in how the OS life-cycle works.

andygup, 2018-05-03:
Conceptual code for a possible fix:

```java

    public void stopLocation(){
        if(_phoneStateListener != null && _telephonyManager != null){
            _telephonyManager.listen(_phoneStateListener, PhoneStateListener.LISTEN_NONE);
        }

        if(_signalStrengthListener != null && _telephonyManager != null){
            _telephonyManager.listen(_signalStrengthListener, PhoneStateListener.LISTEN_NONE);
        }

        _signalStrengthListener = null;
        _phoneStateListener = null;
        _telephonyManager = null;

        try {
            Thread.currentThread().interrupt();
        }
        catch(SecurityException exc){
            Log.e(TAG, exc.getMessage());
            sendCallback(PluginResult.Status.ERROR,
                    JSONHelper.errorJSON(CELLINFO_PROVIDER, ErrorMessages.FAILED_THREAD_INTERRUPT()));
        }

        Log.d(TAG, "Stopping cell location listeners");
    }

```

commit cad27eec (2018-05-15): Closes #61
closed 2018-05-15 by andygup