Esri/cordova-plugin-advanced-geolocation #64: is-it possible to use advanced-geolocation plugin in the background

spinella1anthony, 2018-06-25:
hello everyone, 

Is-it possible to use this plugin in the backgroud? 

Thanks in advance

andygup, 2018-06-25:
This plugin isn't designed to run in the background. Because of its potential to consume large amounts of memory and CPU cycles it will only provide locations, by default, while the application is in the foreground and active.

spinella1anthony, 2018-06-26:
Thanks for your answer.
Is there an other way to obtain SatelliteData in background?

andygup, 2018-06-26:
If you want you can make a copy or fork this repo and customize the functionality for the `onStop` and `onPause` listeners: https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/AdvancedGeolocation.java#L431-L439

andygup, 2018-07-02:
Closing, feel free to reopen if necessary.

closed 2018-07-02 by andygup