Esri/cordova-plugin-advanced-geolocation #65: Need cellinfo while location position is disabled

Marcophono2, 2018-06-30:
Hello!
I read in an older thread that a disabled location positioning by the user also disables access to the network informations for security reasons. My question: Does that also include the cellid information? Out of the box your great plugin does not provide the cellid as soon as I disable the location service (Android 8.1) but may be there are certain settings in the plugin code which could seperate the cell info access from the network info access?

Best regards
Marc

andygup, 2018-07-02:
@Marcophono2 Yes, this plugin still maintains a security check to verify if the [`NETWORK_PROVIDER`](https://developer.android.com/reference/android/location/LocationManager.html#NETWORK_PROVIDER) is enabled in [`LocationManager`](https://developer.android.com/reference/android/location/LocationManager#isProviderEnabled(java.lang.String)). You can find that boolean check here at [advancedGeolocation.java#L227](https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/AdvancedGeolocation.java#L227).

I'm going to continue to leave that verification check in this repo because it represents a security best practice, however you can remove the check yourself in your own copy.

Marcophono2, 2018-07-02:
Thank you, Andy!

closed 2018-07-02 by Marcophono2