Esri/cordova-plugin-advanced-geolocation #66: NullPointerException while trying the docs sample

JCKodel, 2018-09-19:
Only by using the plugin, right after the permission dialog popups and I allow the use of GPS:
```
09-19 18:08:52.755 14999 14999 D CordovaActivity: Resumed the activity.
09-19 18:08:52.756 14999 14999 W com.facebook.appevents.AppEventsLogger: activateApp events are being logged automatically. There's no need to call activateApp explicitly, this is safe to remove.
09-19 18:08:52.756 14999 14999 D GeolocationPlugin: onResume
09-19 18:08:52.756 14999 14999 D GeolocationPlugin: validatePermissions()
09-19 18:08:52.759 14999 14999 D GeolocationPlugin: Counter: 0, display rationale? false
09-19 18:08:52.759 14999 14999 D GeolocationPlugin: rationale 4: application startup - no perms are set yet
09-19 18:08:52.763 14999 14999 D AndroidRuntime: Shutting down VM
09-19 18:08:52.764 14999 14999 E AndroidRuntime: FATAL EXCEPTION: main
09-19 18:08:52.764 14999 14999 E AndroidRuntime: Process: br.com.esx.profix, PID: 14999
09-19 18:08:52.764 14999 14999 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {br.com.esx.profix/br.com.esx.profix.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.lang.String.equalsIgnoreCase(java.lang.String)' on a null object reference
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3454)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3494)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1546)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:102)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:154)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:6186)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
09-19 18:08:52.764 14999 14999 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean java.lang.String.equalsIgnoreCase(java.lang.String)' on a null object reference
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at com.esri.cordova.geolocation.AdvancedGeolocation.startLocation(AdvancedGeolocation.java:254)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at com.esri.cordova.geolocation.AdvancedGeolocation.validatePermissions(AdvancedGeolocation.java:208)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at com.esri.cordova.geolocation.AdvancedGeolocation.runAction(AdvancedGeolocation.java:127)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at com.esri.cordova.geolocation.AdvancedGeolocation.onResume(AdvancedGeolocation.java:423)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at org.apache.cordova.PluginManager.onResume(PluginManager.java:264)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at org.apache.cordova.CordovaWebViewImpl.handleResume(CordovaWebViewImpl.java:450)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at org.apache.cordova.CordovaActivity.onResume(CordovaActivity.java:275)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1270)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.Activity.performResume(Activity.java:6788)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3431)
09-19 18:08:52.764 14999 14999 E AndroidRuntime:        ... 8 more
```

andygup, 2018-09-20:
That error says the `providers` argument wasn't set in the `start()` method. Double check and make sure you initialized all the start options:

```
    ////////////////////////////////////////////
    //
    // REQUIRED:
    // These are required Configuration options!
    // See API Reference for additional details.
    //
    ////////////////////////////////////////////
    {
        "minTime":500,         // Min time interval between updates (ms)
        "minDistance":1,       // Min distance between updates (meters)
        "noWarn":true,         // Native location provider warnings
        "providers":"all",     // Return GPS, NETWORK and CELL locations
        "useCache":true,       // Return GPS and NETWORK cached locations
        "satelliteData":false, // Return of GPS satellite info
        "buffer":false,        // Buffer location data
        "bufferSize":0,        // Max elements in buffer
        "signalStrength":false // Return cell signal strength data
    });
```

andygup, 2018-09-25:
I'm going to go ahead and close this issue for now. Feel free to reopen.

closed 2018-09-25 by andygup