Esri/cordova-plugin-advanced-geolocation #68: ANR issue onDestroy?

junbonglee, 2018-11-29:
[image]

Thank you for the great plugin. I'm having an ANR issue, and the image is screenshot of related ANR from Google Play Console.

I think it's shutdownAndAwaitTermination function causing the ANR, and wondering if terminating threadpool onDestroy is neccessary. If it's not, Is it ok to comment out 

shutdownAndAwaitTermination(_cordova.getThreadPool());
_cordovaActivity.finish();

these two function inside onDestroy()?

I am worried if commenting out these two will cause another problem.

andygup, 2018-11-29:
@junbonglee I'd say go ahead and comment it out and test it. Reference: https://github.com/Esri/cordova-plugin-advanced-geolocation/blob/master/src/com/esri/cordova/geolocation/AdvancedGeolocation.java#L446

The general reason that's code is in there is because of multiple issues where Cordova's shutdown code didn't work properly and probably still doesn't handle internal thread pooling very well. Cordova used to not correctly checking for `Thread.currentThread().isInterrupted()` and that's also caused a variety of issues with other plugins, reference: https://github.com/Esri/cordova-plugin-advanced-geolocation/issues/49#issuecomment-384719552

andygup, 2019-03-15:
I'm closing this out, if something else comes up feel free to ping me.

closed 2019-03-15 by andygup