Esri/cordova-plugin-advanced-geolocation #72: Old location data provided

carliblaz, 2019-03-28:
Hey, I have this strange behavior on Android, if start location stream result - I check if there is GPS location and then I stop the stream. There is this strange pattern in 3 of 4 requests I get old GPS data.

For example:
1 request - old data
2 request - old data
3 request - new data
4 request - old data

Did I miss any settings or what could be wrong?

andygup, 2019-03-28:
What is your configuration `providers` property set to? Reference: https://github.com/Esri/cordova-plugin-advanced-geolocation#use-cases

How are you determining the gps data is old or new? Can you provide an example?

carliblaz, 2019-03-28:
How? By coordinates :)

I'm doing some tests with this cache option.

andygup, 2019-03-28:
What is your configuration for the providers property in this library? 

What make and model of phone?

dslima90, 2019-04-08:
I am experiencing the same issue. A few of the people who installed an app i deployed are reporting this issue... In my cellphone, where i tested, it works ok, though. @carliblaz, how are your tests going?

andygup, 2019-04-08:
I'll need **all** the following information to help with troubleshooting:

- the providers property configuration for this library? For example:

```
{
        "minTime":500,         // Min time interval between updates (ms)
        "minDistance":1,       // Min distance between updates (meters)
        "noWarn":true,         // Native location provider warnings
        "providers":"GPS",     // Return GPS, NETWORK and CELL locations
        "useCache":false,       // Return GPS and NETWORK cached locations
        "satelliteData":false, // Return of GPS satellite info
        "buffer":false,        // Buffer location data
        "bufferSize":0,        // Max elements in buffer
        "signalStrength":false // Return cell signal strength data
    }
```

- Make/model and Android version of phone(s) experiencing the issue
- A console dump of the jsonObjects in the `AdvancedGeolocation.start()` callback that demonstrates the issue

carliblaz, 2019-04-08:
> I am experiencing the same issue. A few of the people who installed an app i deployed are reporting this issue... In my cellphone, where i tested, it works ok, though. @carliblaz, how are your tests going?

Well cache is there for returning cached location results, that was my problem.

dslima90, 2019-04-08:
> Well cache is there for returning cached location results, that was my problem.

Yes, my problem was that sometimes i was receiving the cached location from GPS AFTER i received a new location from NETWORK, and i didn't notice that this was the issue... I guess that was a problem on my side, i just have to filter the results i get using provided timestamps... Sorry for taking your time...

closed 2019-04-08 by carliblaz